sriharshaarangi/BatteryChargeLimit #99: Support for Samsung Galaxy S5 SM-G900T

Kramer18, 2018-03-21:
I've tried all the control files loaded with the program and none are working.

Kramer18, 2018-03-21:
With "batt_slate_mode" and no charger connected it currently says Status: Charging. With the charge connected, it will keep charging past the charging limit set, even while it says Status: Discharging

milux, 2018-03-21:
First of all, we cannot and will not deal with all the possible kinds of device problems. If Samsung decided to screw up your ROM... bad luck, sorry, nothing we can do here.

Anyway:
Stock ROM or custom ROM?
Any magical battery saver apps or other stuff running in the background?
Any special charger, or the default one?
Where are our logs according to the FAQ on XDA?
Anything else we may need to know that could potentially screw up your setup?

Kramer18, 2018-03-21:
Milux- Thanks for your reply!!!

Stock ROM

I am using an Anker charging cable that has a lower amp output than stock.

I have PowerPro installed, but I forced stopped it prior to running Battery Charge Limit last.

How do I send you the logs?

I have BitDefender and Security Master installed. 

Very odd experience running Battery Charge Limit. Won't start charging until I toggle the Control File option. Then, it seems to work fine for a while. A few minutes after it reaches the charging limit and is in the discharge phase, it begins charging again even though the status still says Discharging. The red charging light on my phone illuminates in contrast to the normal charge phase, when it was NOT LIT. I have to unplug the cable to stop charging, but when I do, the battery meter/gauge continues to rise in %.

Hope you can help and thanks for your efforts.

Daniel

--------------------------------------------

 

 
 
 
 
 
 
 it on GitHub, or mute
 the thread.

Kramer18, 2018-03-22:
- 03-21 21:45:37.880 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:45:52.889 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:46:05.772 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:46:08.479 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:46:16.864 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:46:21.860 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:46:24.050 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:47:01.365 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:47:01.835 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 1   
03-21 21:47:36.065 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:47:56.007 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:48:09.857 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:48:20.362 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:48:24.094 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:48:33.858 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:49:36.881 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:49:47.881 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:50:25.802 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:50:54.860 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:51:06.850 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:51:25.928 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:51:46.860 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:51:51.882 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:03.535 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:08.859 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:12.858 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:42.851 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:52.869 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:52:58.091 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:53:02.555 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:53:02.945 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 1   
03-21 21:53:52.635 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:53:57.065 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:54:00.881 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:54:11.598 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:54:34.861 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:54:52.849 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:55:38.104 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:55:52.870 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:56:07.862 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:56:38.227 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:56:51.862 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:57:12.862 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:57:24.259 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:57:40.940 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:01.575 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:02.195 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 1   
03-21 21:58:14.225 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:24.861 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:33.872 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:44.310 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:58:52.863 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:59:05.452 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:59:09.311 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:59:37.873 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 21:59:59.894 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 22:00:32.855 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 22:01:08.854 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
03-21 22:02:06.276 I/qdhwcomposer(  432): handle_blank_event: dpy:0 panel power state: 0   
[... cut ...]
   -    

   - Odd things I notice:   

   - 1) Phone's charge light is not on while charging, but is on after I disconnect the cord. At that point the Status says "Charging" even though it is not (cord disconnected)
   - 2) Several minutes after the charge limit is reached (with the cord connected), the Status continues to say Discharging, but the phone resumes charging. No charge light. After watching the battery gauge % continue to rise, with an accompanying rise in voltage, I pull the plug. See #1

leo644, 2018-07-24:
i have the same problem, what can i do to resolve it?
maybe using another ROM or something like this?

milux, 2018-10-02:
This looks pretty much like the ROM or the hardware screws up things here, or we are simply using the wrong control file. (or some entirely different reason...)
Sadly, the logs given so far contain no useful information at all. 😞