sriharshaarangi/BatteryChargeLimit #104: hidden charge state in OnePlus 5T

ZoharML, 2018-04-02:
There is some kind of a bug in the 5T roms that allows it to take charge from the USB cable even when "charging_enabled" = 0
When this happens the charge icon will not change to charge and the battery % will not get higher!
it is possible to recognize this state according to \sys\class\power_supply\battery\current_now that will change to negative values. at this state changing the value of /sys/class/power_supply/battery/input_suspend to 0 and then back to 1 will stop this hidden charge state.

This state will only show up from time to time. I did not found the logic after it.

More info can be found at:
https://forum.xda-developers.com/showpost.php?p=76050666&postcount=1770
https://forum.xda-developers.com/showpost.php?p=76102724&postcount=1774

Thanks for your work,
Zohar

milux, 2018-05-15:
Changed categorization to "future work", because it will require checking another file, which is
a) likely not available across all devices.
b) could cause other unforeseen consequences and weird bugs with devices that have again a completely different logic for this matter.

sriharshaarangi, 2018-05-15:
@ZoharML , as you stated in https://forum.xda-developers.com/showpost.php?p=76158248&postcount=1776, there is no definite solution to this problem right?

ZoharML, 2018-05-17:
I could not find a "cure" that will fix the problem at 100% of the time.
I did find the control files that can detect that the phone is in this state (at least for the 5T).

What relay fixed the problem for me was to stay away from my old charger.
It was a "travel adapter" model MCS-04WR, output 5V@1A, it came with my old LG phone.

sriharshaarangi, 2018-05-17:
Since this is not an issue with app, and there is no solution to this problem, I will be closing this issue.

closed 2018-05-17 by sriharshaarangi