sriharshaarangi/BatteryChargeLimit #106: Asus Selfie Control File

petepaul1228, 2018-04-13:
/sys/class/power_supply/usb/present
Enable=1
Disabl=0

milux, 2018-04-15:
Have you checked correct functionality via Terminal Emulator?
Afaik a file with identical or very similar name exists for numerous other devices, but does not influence the behavior of the charging circuits.

petepaul1228, 2018-04-16:
Yup! I already set it in control file and it works.
This also works:
/sys/devices/soc.0/78d9000.usb/power_supply/usb/present

The only problem with it is when i disconnect the charger, the charge icon will not go away.

milux, 2018-05-02:
This is normal for some control files, as the OS cannot notice the change until reactivation of the charging component.

petepaul1228, 2018-05-03:
Thanks! Kudos!