sriharshaarangi/BatteryChargeLimit #107: App not closing properly/continues to use root access when power is disconnected.

neonbulb, 2018-04-17:
When I plug the phone in to charge it says "battery limit service enabled"
charges up to to 80% and all works good.

when I unplug the charger it says "battery limit service disabled" (all good)

Problem: If i go check which apps are using root it says the app is running even though my phone is not plugged into a charger. On my status bar it continues to show the "#" icon meaning that an app is currently running using root.

Solution: If I open the app then close it, it will stop running and stop using root access. 

It seems to be as if the service is disabled but the app never actually closes so it continues to use root access?

This bug is occurring on LineageOS

sriharshaarangi, 2018-05-14:
The app doesn't have a background service. The app is active only as a foreground service (as long as the notification is present). You can check in _Running services_ under _Developer options_, and the app won't be present after _Battery Limit Service Disabled_ toast.

milux, 2018-05-15:
I can still understand what @neonbulb is talking about.
Actually, this is kind of a "it's not a bug, it's a feature" case.
The app has a helper class that opens a interactive su shell that remains active until the OS kills the app to free resources.
This is because creating a su shell for command execution can take a moment, which renders the app somewhat unresponsive in certain cases.
I hope that this will improve as soon as we reduce the usage of the root shell as suggested in https://github.com/sriharshaarangi/BatteryChargeLimit/issues/47