sriharshaarangi/BatteryChargeLimit #114: Pixel XL issues

droid-jd, 2018-05-15:
I have a pixel XL and I've noticed a couple of bugs.
I use the "battery_charging_enabled" control file, which I believe is the correct file for this device. I also have my phone set to 80/70.
I'm running version. 1.1.1, but I've had the overcharging issue since earlier versions as well..

Bugs:
1. I'll plug phone in overnight, and when I take it off the charger it reads 80%, as expected. The problem is that it'll continue to read as 80% for about 20 monies of solid use, leaving me to believe that the device is really charged to 100, but only displaying 80%. 

2. When using the app, my phone will not discharge while in the charger. By that I mean, it charges up to ,80%... Then stays at "80%" until I take it off the charger.

milux, 2018-05-15:
Regarding 1.: Have you checked whether the voltage information displayed in the app activity confirms this assumption? If your assumption is correct, then you will see peak voltage between 4.2 and 4.4 V in the morning. In this case, your device might not behave correctly indeed.

Regarding 2.: Congratulations, that's exactly what should happen with this control file. The battery charging is stopped, but the phone will get required energy from the AC adapter as long as it is attached. That's best case behavior, no bugs here... ^^

droid-jd, 2018-05-15:
Ahh. So now things get even more confusing:

1. In the notification shade, the app reports 4.4v. but in the app it reports 4.159. Also, upon unplug and replug of the charger, the app reports v4.399. after a few monies though, it went back down to 4.165

2. Kinda reiterating 1, but if I wanted the app to charge/discharge between the set values, do you know which control file I would use? I read through the battery information you posted in the xda OP, and I understand the A/C draw is more beneficial to battery longevity, but to satisfy my own curiosity, I'm wondering how I would configure such a feature?

sriharshaarangi, 2018-05-17:
The one in the app is instantaneous and the notification value is an outdated one(bug, we need to fix this).
1. I guess the phone is charging to 100% according to the voltages :( even though it is reporting 80%
2. If there is another control file, try it and report. 'charging_enabled' is one such control file which allows you to discharge and charge between the set values.

droid-jd, 2018-05-17:
Awesome
Thank you for your help!
Even though the problem that I had was mostly user error, I am glad I could at least help identify one bug (the notification value).

closed 2018-05-17 by droid-jd

sriharshaarangi, 2018-05-18:
@droid-jd 
Why did you close the issue? Is your issue solved? Were you able to find an alternate control file which works as expected?

droid-jd, 2018-05-18:
Yes, the "charger_control" file works the way you described it should. I tried the other ones available, but they didn't function properly.
It charges to my desired level (80%), then enters "discharge" mode, but still holds at 80. The voltage also stays around 4.1, I confirmed with Ampere.

Previously, i didn't try different control files based off something I read in a different XDA thread, saying the pixel XL used the "battery_charging_enabled" control file.  Had I just followed through with the OP directions, and not what I thought I knew, this would have been resolved before I started questioning it.

Thank you both again for your help!

milux, 2018-05-18:
Interesting... how can it work for all the others, but not for you? 🤔 
Are you using some unusual custom ROM?

droid-jd, 2018-05-18:
Nothing unusual. Running lineage 15. 
Perhaps l wrongfully assumed that because it's AOSP based, it would use the typical/default pixel charge file.