sriharshaarangi/BatteryChargeLimit #117: Why not 100%?

Freeedim, 2018-06-06:
Hi and thanks for this great app! But why isn't it posso le to set the end of the charging when the battery reaches 100%? Wouldn't it be better for the calibration?

milux, 2018-06-19:
If you want your device to be charged up to 100%, just disable BCL!
If you send an Intent or a Broadcast with 100, BCL will be disabled equally.

closed 2018-06-19 by milux

Freeedim, 2018-06-20:
Well, after a lifetime charging various devices, I had figured out that I could reach 100% if I just let the charging process go on. Thanks for the advice. What I try to achieve now is reaching 100 % then immediately stopping the charge in an automated way.

milux, 2018-06-26:
I thought so. xD
But seriously, I don't really see why this makes sense, can you please elaborate?
When charged up to 100%, the device won't charge the battery any further anyway. (Well, at least approximately...) Letting the device discharge and then charge up to 100 % does more harm then good, so what would be the point of this?

Freeedim, 2018-06-28:
What it used to do: the battery is charged until 100% then it is kept there by slightly discharging and recharging as long as the cable is plugged.

What it does now with BCL: the battery is charged until n% (n being 99 or less) then it will not be recharged again until the minimum threshold is reached, even if you leave the cable plugged.

What I would like it to do : same sentence as previous except you replace "99" with "100".

sriharshaarangi, 2018-07-14:
Many people have asked me the same thing, to allow them the limit set to be 100%. I have changed the max limit to 100% in v1.1.1, and people who want to use it can use it.
(@milux Also, I'm seeing some people who are setting lower limit at around 40, and upper limit in 80s. 
To avoid such settings, I'm thinking to limit values of upper threshold and recharge threshold, and add a switch in Advanced Settings to enable full limits, saying do this if you completely understand what you are doing)