sriharshaarangi/BatteryChargeLimit #119: Delay charging

thabubble, 2018-06-21:
A good feature would be to be able to delay charging for x amount of hours and /or see the time when to start charging

milux, 2018-07-11:
If you're talking about a time delay between hitting the upper limit and recharge - agreed, this would be a useful feature.
If you're talking about a general setting to delay charging:
This is outside of the scope and dev goals of BCL. However, you can easily combine BCL with Automate or Tasker to send limits to BCL using Broadcasts. This way, you can implement such any time- or state-controlled behavior. Indeed, I'm using Automate myself to increase the charging limit from 65 % to 80 % between 4 am and 6 am - works like a charm!

paulie-g, 2019-04-03:
A charging scheduler would be very useful and I would urge you to reconsider.

Here is a use case: an older family member, despite being told over and over that he should not be leaving his devices plugged in for 12 hours straight (phone) or days on end (tablet), continues to do so. It is not within my, or your, power to solve this. I could, however, if scheduling functionality were available, set it up so that BCL knows that he needs to have 100% charge at a given time every day (in case he wants to take the device with him to the office) and keeps it between, say, 60% and 80% the rest of the time. I can teach him to disable BCL for one-off cases, but he won't remember to do it every morning any more than he currently doesn't remember to just plug it in for an hour every morning to charge.

Yes, this can be accomplished with tasker/automate. Heck, I don't even need BCL at all - now that I know what the control files are, I can just run a shell script from init.d or cron. This is not, however, particularly user-friendly. Scheduling functionality is useful and directly related to the function of BCL - limiting wear and tear on the battery (in this case, while still providing the user with enough charge at the right time to keep the device useful throughout the day).

All it would need is a secondary screen with a selector for hours:minutes, a checkbox for whether that's a start or end time, and an optional target charge level. Most of this functionality could be lifted from one of the open source alarm clocks. If it didn't take me days just to get the toolchain to work to build the last Android project I wanted something fixed in, this wouldn't be a comment but a pull request. As things are, for me personally the time/reward ratio is such that I would probably just do it all in a shell script in 15 minutes instead - especially if you think it's out of scope and wouldn't merge it anyway.

Imo, the biggest value of BCL relative to the other projects that do similar things is precisely that it provides a GUI, and a way for grandpas/grandmas to disable the service on a one-off basis to charge to 100% without opening a terminal. Thank you for all your work by the way - I don't mean any of this to come across as entitled or ungrateful, I'm very appreciative of all the work you've put in and am profoundly embarrassed that I'm even asking you to implement this instead of submitting a PR. It's just an incredibly unrewarding timesuck to re-learn all the braindead unnecessary complexity of Android dev every 2 years when I want to patch something.