sriharshaarangi/BatteryChargeLimit #128: Allow user to change notification priority

Cubox-, 2018-09-06:
Hi,

Thanks for the great app, I rooted my phone primarily to use it.

I see that you are active and still updating the app, and I would like to ask a small feature to be added.

Could we have an option to pick the priority for the notification while the service is running?
I'm assuming that the notification is needed to keep the service alive, and I don't want to block it completely.
Having the priority as minimal would be perfect for me!

Thanks for your time on this open source great app, have a good day

sriharshaarangi, 2018-09-07:
Yes, I have noticed it recently as well, and wanted to change that. That part of code goes back to when I was new to Android development. Thanks for the feedback!

milux, 2018-10-02:
> Yes, I have noticed it recently as well, and wanted to change that. That part of code goes back to when I was new to Android development. Thanks for the feedback!

Actually I did several changes there when I started developing with you in that project. A.t.m. the priority should be as high as can be, because numerous users reported the app being killed by the OS, and I really got sick of dealing with those requests...