sriharshaarangi/BatteryChargeLimit #129: Non root mode

zputnyq, 2018-09-16:
Hi @sriharshaarangi.
I'm new battery charge limit user.
I love this app & I use rooted device.
Actually it's just a feature request.
I believe this app will be greater if it has non root mode, without modifying system file. So it will be only a notification when battery charge reach x%. Probably it won't stop charging but sometimes it's enough for some people.
Thanks, I really appreciate it

milux, 2018-10-02:
Thinking a moment long about it... isn't that functionality already available? 😕 
There's a "notify when limit reached" switch in the main view, and recent versions should not terminate when they can't detect root, do they?

zputnyq, 2018-10-02:
So this app actually could work without root ?
I am confuse here

milux, 2018-10-02:
I would hope so, but never checked it out explicitly.
Feel free to do some experiments and report back! 😉

zputnyq, 2018-10-02:
After few hours of experimenting.

1. This app fc when its root permission is denied.
2. "Notify when limit reached" is unable to work when "Enable battery charge limit" is inactive eventhough "notify when limit reached" is activated.

sriharshaarangi, 2018-10-05:
@milux The notify feature works on top of the normal functionality. I haven't decoupled the non-root part.

milux, 2018-10-15:
Alright, then this should be a top-priority for enhancement of BCL, thanks @zputnyq.

zputnyq, 2019-09-26:
Hello @sriharshaarangi & @milux 
I have an idea but of course if it's possible to be implemented.
It is about delaying charge to cool down temperature. So on a specified temperature charging will be delayed for xx seconds to cool down, then after it's cool down it will automatically start again. I think this will be great.

jmichael2497, 2020-02-10:
> only a notification when battery charge reach x%

you should always use complete and specific subject titles.

you can edit your OP to fix to subject title to be more accurate and specific to this request like "Non root mode" to "non-root mode for charge limit notification only".

maybe it will encourage someone to help work on implementing it.

zputnyq, 2020-02-11:
> you should always use complete and specific subject titles.
>
> you can edit your OP to fix to subject title to be more accurate and specific to this request like "Non root mode" to "non-root mode for [...]

Not necessary it's almost 2 years now & I don't wanna encourage someone else at that time, only bcl team because it's their app. They wanna implement it or not it's up to them

jmichael2497, 2020-02-16:
> Not necessary it's almost 2 years now

and yet i see a few months ago you added an update with an unrelated feature request, instead of checking for existing similar requests, before opening a new issue.

@zputnyq commented on Sep 25, 2019
> Hello @sriharshaarangi & @milux
> I have an idea but of course if it's possible to be implemented.
> It is about delaying charge to cool down temperature. So on a specified temperature charging will be [...]

maybe you're nor familiar with how https://guides.github.com/features/issues are supposed to be used, but... searching before making a separate, accurately titled, described, and revised-as-needed OP are the best way to help everyone track them and avoid creating duplicates... as well as closing your own issue if/when it is resolved.

hope that helps, cheers.

zputnyq, 2020-02-17:
Ok that could help

Klimmbimmel, 2022-04-06:
> Alright, then this should be a top-priority for enhancement of BCL, thanks @zputnyq.

This feature is probably offered by a lot of other battery tools available on F-Droid