sriharshaarangi/BatteryChargeLimit #132: TECNO Camon CX Air Support Request

mikesew1320, 2018-09-25:
There is No charge enabled disabled option on set control file menu only moto defy+ w/c is experimental available.Please make it to support my device,I can send you whatever you want from my phone.just contact me!

My Device is:-
                       BRAND-TECNO
                       MODEL-CAMON CX AIR BUILD V141
                       ANDDROID-7.0 AOSP HIOS2.2.0
                       SOC-MT6737    
                       RAM-2GB

I hope you will accept my request and add it on next update!

                  **********THANKS**********

mikesew1320, 2018-09-25:
If there is a method to add my control file to this app manually  please help me!

milux, 2018-10-02:
There is, have a look at the settings please!
We need a control file that is **tested** to work on your device, otherwise support will not be available.
Please look at other issues and the XDA channel for similar requests, I don't want to repeat myself over and over again. ;)

mikesew1320, 2018-10-02:
Where can I get the control file and test it on my phone?

milux, 2018-10-15:
If this really matters for you, then you will dig it out in the XDA thread. Sorry, but we don't have enough time to handle this kind of requests. This issue can be re-opened when you have new information for us.

mikesew1320, 2019-04-27:
please refer here and help me @ https://github.com/sriharshaarangi/BatteryChargeLimit/issues/76
I have got my phone control file @ /sys/class/power_supply/ac/online
What I need is your help to fix the problem I faced @#76  to test it via Terminal Emulator.

If you can please reopen the issue and reply to me I want your app to work on my phone!

mikesew1320, 2019-04-28:
I found my charging control file @ /system/class/power_supply/battery/status
Enabled Value.---  Charging
Disabled value--- Full
I am not Testing it due to the issue I faced on Terminal Emulator @ #76 and also it can't change the value of the status file @ chat locatiekeuze terminal emulator I filmpjes
su
Cat........
echo Full ... 

But when I see after running the command using cat....... it is still Charging 

Please help me

mikesew1320, 2019-04-28:
I found my phones control file @ /sys/class/power_supply/battery/status

Enabled---- Charging
Disabled--- Full
When I unplugged it---- Not charging

mikesew1320, 2019-04-28:
I also triest to set  a custom control file using your app V 1.1.1 but it only notifies me but not stopped my charge. Remember that I enabled Always write control file on Setting.

I noticed that you are not active on github. Please replay to me

milux, 2019-07-13:
> I also triest to set a custom control file using your app V 1.1.1 but it only notifies me but not stopped my charge. Remember that I enabled Always write control file on Setting.
> 
> I noticed that you are not active on github. Please replay to me

I didn't reply because your findings indicate with high probability that your "control file" is actually only a status indicator and not working as a real control file. Sorry.

mikesew1320, 2019-07-14:
Sorry for my last request, I found my phone control file at "/sys/devices/platform/battery_meter/FG_daemon_disable"

I tested it using terminal,magisk modules Acc and IPC also by using your app as a custom charging control file.

The only issue with this control file is it stack the battery percentage to the top off value untill a reboot, I think it's very useful if you add rebootOnUnplug feature for such like charging control files.

Thank you!

milux, 2019-07-16:
Reboot on unplug doesn't sound like a feature that many people would like to have...
But OK, it's a valid control file, so that justifies to reopen.