sriharshaarangi/BatteryChargeLimit #134: Android wear support

sooti, 2018-10-18:
Hey any chance of getting support for Android wear, I just replaced my battery for my LG Urbane w150 and wouldn't want it to wear out so fast, thanks!

milux, 2018-10-20:
Sorry, I have no hardware and no experience about development for Android Wear, so if you don't do it yourself, then most likely not.

sooti, 2018-10-23:
I actually got it working on wear 2.1, just did adb install on my rooted watch, seems to work fine, if you could somehow add a wear layout in your layouts  (like this https://developer.android.com/training/wearables/ui/layouts), that would be great, thanks anyway!

milux, 2018-10-25:
Good to hear that, but I do neither have a watch for testing, nor any experience in designing good watch layouts, so I will not be able to develop that layout.
Therefore, we're going to wait until somebody develops this layout and does a pull request.