sriharshaarangi/BatteryChargeLimit #137: Illusion of limit (Meizu 15 Lite).

JulyMorning, 2018-11-02:
Today charged my phone all night and when I woke up it was 75%, as I set it to be. However, it lasts for about 2 hours. I've even uninstalled and rebooted the device.

It was used with default options, the only thing was using some kind of delay (enforcing device to recognize changes) and handling power events.

[image]

milux, 2018-11-05:
Maybe your device is continuing the charging process, even though it says it doesn't. On the other hand, it could also happen that the algorithm doing the battery status tracking is simply "confused".
What is your battery voltage in the morning, some seconds **after disconnecting the charger**?

JulyMorning, 2018-12-07:
I've noticed that the problem occurs only when I charge the device with my notebook which has a powerbank-like feature: it allows to charge devices being turned off. When I charge it the other ways, I guess, it stops charging correctly – at least, I see that the charge became lower pretty soon.

Having limit = 80%, I have 4.079V when I disconnect it from my regular power source. I'll try to charge with the notebook for half an hour and will report the numbers then.

Does the control file affect the charging process? qpnp-smbcharger-18 seems to work for me too. is it better in any way? Now I use universal mode, but one month ago I guess it was battery_charging_enabled.

JulyMorning, 2018-12-07:
Obviously, I was wrong: it continues to charge no matter what source of power I use.

JulyMorning, 2019-04-11:
Using qpnp-smbcharger-18, I found 2 things. When `_battery_charging_enabled_ = 0,` but power plugged in, I have:

> battery status: discharging
> power plug: USB (or AC, depending on source)

battery stops being green inside, but there's still voltage icon on the left.

when I manually set _battery_charging_ to 0 (the same folder; _/soc/qpnp-smbsharger-18/power_supply/battery/_), it actually stops using the cable, no «power plug» status, only discharging. And I can actually see the discharging process still being plugged in.

I will double check if the device overcharge itself despite having the precentage number unchanged using the first (default) setting, but if the app allows to set the upper and higher limit to maintain a battery's charge within these numbers, only the second approach seems to be working, at least with my device. The first one just doesn't allow it to discharge.

Is it possible to make an advanced feature that allows several ctrl files to be changed? Just to be sure. Also I' interested in 2 files in that folder: voltage_max and voltage_max_design. The first file has 4400 inside and the second has 4400000. I've actually seen my battery's voltage going that high, to ≈4.4v, when 100%. Is it safe to change them? Thanks.

milux, 2019-07-13:
@JulyMorning The recent beta version in the Telegram beta tester group includes an experimental feature for voltage control via one of the files you mentioned. You may want to test this.
So those files would be control files, right?
```
/soc/qpnp-smbsharger-18/power_supply/battery/battery_charging_enabled
/soc/qpnp-smbsharger-18/power_supply/battery/battery_charging
```
Which of those do actually stop the charging process?
If the voltage of the battery stays constant with the former, then this would be the ideal behavior, and you may want to use that one.