sriharshaarangi/BatteryChargeLimit #139: Found control file? Razer Phone 1

room312, 2018-11-13:
- **/sys/class/power_supply/usb/device/razer_charge_limit_enable** – 1 or 0. Enables or Disables the charge limit feature, respectively. 0 is default.
- **/sys/class/power_supply/usb/device/razer_charge_limit_max** – The percentage at which charging stops and discharging begins. 70 is default.
- **/sys/class/power_supply/usb/device/razer_charge_limit_dropdown** – The percentage at which charging begins again when the battery has discharged down to this level. 65 is default.

The values reset back to default after reboot.

(I'm pretty sure root is needed to access/modify these files. I’m rooted on 8.1.0, July 5, 2018 patch. Using Magisk.)

This app, by default, is using /sys/class/power_supply/battery/input_suspend to automatically set the charging state. However, the change was purely cosmetic since the phone kept on charging.

milux, 2018-11-14:
Nice... unfortunately, this requires a lot of semantic changes in BCL, so I tagged it "future work".
Unless somebody implements this and issues a PR, it might take some time until this gets implemented.

milux, 2019-07-13:
@room312 I think I had overlooked something here... the first file is a "classical" control file according to your findings, correct? In that case, we can add it for the next release.
Tagging "todo", sorry for me misreading this...