sriharshaarangi/BatteryChargeLimit #143: Flashlight and vibrations stop working after reaching charge limit

rozwell, 2018-12-03:
On Samsung A5 2016 ( SM-A510F ) when charge limit is reached, flashlight and vibrations stop working.
To fix this, I have to disconnect device from the charger, disable limit from notifications and then plug in the charger for a second, efore charge limit is applied again.
Control file used: `/sys/class/power_supply/battery/batt_slate_mode` (default)

Only disconnecting the device and disabling the app isn't enough, both flashlight and vibrations will not work until device is either charging again or rebooted - those are 2 workarounds I know, setting value to batt_slate_mode manually by echo didn't change anything.
I should add that disconnecting the charger doesn't automagically close the application, it's still in notifications.

If needed, I can provide more informations but maybe it's a known issue and there is a simple workaround?

milux, 2018-12-11:
Hey @rozwell ,
sorry to say that, but this is a device-specific issue, which means it's out of the development scope of this application. Maybe we can include some kind of custom broadcast in future versions so that such issues can be tackled with a device-specific workaround via Automate or Tasker, but for now I don't see any simple solution here, you'll have to go with your manual workaround for now.

> I should add that disconnecting the charger doesn't automagically close the application, it's still in notifications.

This is indeed known behavior, the removal of the charger cannot be detected, unless charging is again enabled, for technical reasons.

rozwell, 2018-12-11:
> Hey @rozwell ,
> sorry to say that, but this is a device-specific issue, which means it's out of the development scope of this application. Maybe we can include some kind of custom broadcast in future [...]
Hello @milux ,
I'll look for a solution once I have more time.
Personally I'm using MacroDroid, are those 2 other any better?

franzalex, 2018-12-12:
@rozwell

I evaluated several automation apps previously and settled on using Automate on all my phones. As compared to the experience I had with MacroDroid and Tasker, I can definitely say Automate is far easier to work with.

closed 2019-02-16 by milux

rozwell, 2019-03-31:
While looking for a solution, I stumbled upon [ACC](https://github.com/Magisk-Modules-Repo/acc/) which seems to have everything covered and no issues reported here. You may find some ideas there.
It just lacks of GUI, which is supposed to be covered by an app called [AccA](https://github.com/MatteCarra/AccA/) but I didn't use it.
I'd really prefer BCL since it's really nicely done, but I can live with a config file which I'll probably won't touch for months.

milux, 2019-05-22:
ACC is just same same but different. ^^
I wouldn't bet that this huge collection of bash scripts has superior stability compared to BCL, and the project is too young to have issues assigned yet.
However, I agree that it looks promising. And the author tries to hack in every single feature you can thing of.
If it works for you - perfect - go for it! :)