sriharshaarangi/BatteryChargeLimit #144: Momentary Stop Charge and Start Charge near the limit

skai59, 2018-12-09:
Just a minor issue, but when my phone reached the limit, it often (not always does, but a majority of times) stops charging (which it should), but then the phone instantly drops by a percent, causing the phone to start charging again, in which it stops again after a few seconds because it has reached that percentage again.

It's a minor issue, but it feels annoying, and maybe a slight delay after reaching the limit will help? Anyways, thank you for developing this app! It has really helped me keep my phone charged up to 85% and preserve its battery. Cheers.

milux, 2018-12-11:
Sorry @skaiteo98, but according to your explanation, you set lower limit = upper limit, which is known to cause exactly that kind of issues.
Please use a difference of at least 1 (better 2, default) percent in order to avoid this kind of cycling, as this high frequency cycling will also impact your battery longevity!

closed 2018-12-11 by milux

skaiteo98, 2018-12-11:
Sorry, I think my explanation has caused a bit of confusion. I **did not** set my lower limit = upper limit. It was set 5% lower than the upper limit.

milux, 2019-02-05:
It's an issue of your phone, nonetheless. Nothing we can do about this on our side. Increase the difference between the two limits until this issue doesn't appear anymore.