sriharshaarangi/BatteryChargeLimit #148: The app prevents Pixel 2 XL from charging when USB-C to USB adapter is used

sdsurfer2015, 2018-12-30:
I used a usb-c to USB adapter to charge the phone but the app prevents the phone from charging even if I disable the charging limit.

milux, 2019-07-13:
What indicates that this behavior is related to the mentioned adapter in any way?
Seems like a ROM-specific bug to me...

sdsurfer2015, 2019-07-14:
I don't see this bug if I plug it directly to a charging port without the
adapter.

milux, 2019-07-16:
I see... so once you charge the phone using the adapter, after charging has been disabled once, it cannot be reactivated?
What happens if you plug it out and back in with the adapter?
I can only speculate that this bug is caused due to different charging modes/currents. In any way, I doubt that we can do much on our side, as this seems to be some device/adapter-specific glitch...