sriharshaarangi/BatteryChargeLimit #150: LG V30 & V30+ support request

solidsneak, 2019-01-05:
The LG V30 and LG V30+ are identical phones aside from the internal storage on the V30 being 64 GB and on the V30+ it's 128 GB. This should be applicable to both.

**file:** "/sys/class/power_supply/battery/battery_charging_enabled"
suggestion for **details:** "LG V30/+"
**chargeOn:** 1
**chargeOff:** 0

Please let me know if you need more information.

milux, 2019-07-13:
This file is already available, tagged with "Nexus" (the phone were it was observed first).
I don't see the point of this request. If I'm mistaken, please reply to me and I will reopen.

closed 2019-07-13 by milux

solidsneak, 2019-07-13:
The point was in the parenthesis in the title.

milux, 2019-07-13:
@solidsneak Good point, thank you. As some control files work for a large variety of devices, we usually stick to the first device where it was observed to work, but maybe this will be changed in the future.