sriharshaarangi/BatteryChargeLimit #161: Vernee Apollo - Not Working! Why? Please HELP! Thank You!

Exetreezz, 2019-03-15:
Hi.

I am using "Battery Charge Limit 1.1.1" app for a few days now on a "Vernee Apollo" phone (the 2K version not the "Lite" version) which runs Android 6.0. The phone has full root.

My issue with this app is that I really don't understand this:

Doesn't suppose for this app to stop charging the phone when it reaches the exact limit I set inside its options?
No matter what limit I set (70, 80, 90, 95, 97, 98, 99) after it is reaching that limit I get a notification from the app that the limit has been reached out but the phone is still charging up to fully charge (100%) and I don't understand why is behaving this way.

Do I need to set anything else in its settings for your app to stop charging my phone once the specified limit has been reached?

I really need your help in this matter, please!
Thank you so much for your time!

P.S.: I have to mention that I don't have programming skills or something like this, so ... "take me easy" for solving this, please! Thank you! :)

milux, 2019-05-08:
Sorry but we can't help here, our time is limited. If it doesn't work with available control files, you have to dig yourself.

closed 2019-05-08 by milux

Exetreezz, 2019-05-08:
> 
> 
> Sorry but we can't help here, our time is limited. If it doesn't work with available control files, you have to dig yourself.

Thank you.