sriharshaarangi/BatteryChargeLimit #165: Tecno Camon CX Air Support request [Reopened issue] refer #132

mikesew1320, 2019-04-28:
This is a not a new issue for support request for TECNO Camon CX Air, rather it is reopened by me again to get attention.

Refer #132 die the film info please!

milux, 2019-07-13:
Closed, see referenced issue.

closed 2019-07-13 by milux

mikesew1320, 2019-07-14:
Sorry for my last request, I found my phone control file at "/sys/devices/platform/battery_meter/FG_daemon_disable"

I tested it using terminal,magisk modules Acc and IPC also by using your app as a custom charging control file.

The only issue with this control file is it stack the battery percentage to the top off value untill a reboot, I think it's very useful if you add rebootOnUnplug feature for such like charging control files.

Thank you!