sriharshaarangi/BatteryChargeLimit #166: Suggestions for Battery Charge Limit: Add the "ALARMS" feature in the settings section (Main settings)

ammarraisafti, 2019-04-28:
# Components
Stop charging at the desired level to maintain the battery health for the most devices !, my proposal is based on alarm settings.

**The following is a component of the features I want to propose:**

- ALARMS

# Proposal description
Add the "Alarms" feature in the menu settings (main settings).  Alarm overview, this page allows you to add a new alarm, activate or deactivate an existing alarm quickly, or delete an alarm.  To add a new alarm, piss "add alarm" - this will create a new alarm and take you to the alarm settings page, where you can control the settings for the new alarm. 

**Below the "add alarm" button is a list of alarms that you have made.**

- To edit an existing alarm, just press.  
- To delete an alarm, long press, then select "Remove alarm" (You can also delete the alarm from the individual alarm settings page by pressing MENU then "Delete alarm"). 
- To switch quickly whether the alarm is activated, press the alarm clock icon to the left of the alarm text  .

# Mockups / Examples
[image]

[image]

[image]

[image]

# Benefits
You can have BatteryBot notify you when your battery level reaches a certain point (e.g. the battery becomes fully charged,the battery level drop below 20\%). This feature is very interesting to try, and is perfect for everyday use. The presence of this feature will have a major influence on the progress of the application in the future. I hope, my suggestion can be an additional idea for developers to make this application more interesting. 

Thanks.

milux, 2019-07-13:
I know that such an answer is always disappointing, but I cannot see the massive impact of this feature for the relevance of our app.
If you want it implemented in the near future, please write code for it and provide a PR.

RichyHBM, 2020-04-10:
I think something like this would be useful to be able to provide a use to noon root phone, have it alert the user to unplug their phone