sriharshaarangi/BatteryChargeLimit #178: pixel 4 ignores charge limit

Draic, 2019-11-02:
the only control file offered on the pixel 4 is "status Motorola Defy+ (MB526)", but it does not work. Tried it with and without the option to enforce charge limit. Is there something else I need to do here? App always worked fine on Pixel 2 (but had a different control file option there)

JLJTGR, 2019-12-06:
I found a helpful review on the Play Store.
[image]

These settings have worked for me when manually configuring the control file.

heatl0rd, 2020-08-09:
For those that need a copy paste

sys/class/power_supply/charger/charge_disable