sriharshaarangi/BatteryChargeLimit #181: Need help, not working on Realme X.

JoeNirvana, 2019-12-21:
Probably because it use OPPO VOOC charge, any solution?
Kernal source is here:[https://github.com/realme-kernel-opensource/realmeX-kernel-source](url)

marcelopm, 2020-11-19:
Same here, realme 5 pro

milux, 2020-11-24:
> Probably because it use OPPO VOOC charge, any solution?
> Kernal source is here:[https://github.com/realme-kernel-opensource/realmeX-kernel-source](url)

Possible, but BCL (and most similar apps/scripts, I believe) are just changing the content of files that control the charging logic, so nothing we can do here. You can check whether ACC is doing better on your device, maybe there is some control file BCL doesn't know of...

closed 2023-08-16 by milux