sriharshaarangi/BatteryChargeLimit #184: Samsung SM-G389F app works sometimes not during night

Bagh33r4, 2020-01-22:
Hi I got a Samsung Xcover 3 model: SM-G389F. The app works mostly fine during the day, but every night when I have the phone plugged in I always get a fully charged phone at 100% when it should stop at 80%. Please let me know if you need any more information, I have an other Samsung S4 which works perfectly with the app. What else could i try, I have tried changing the control file but then nothing worked. I tried making the phone not sleep during charging, that did not help.

[alogcat.2020-01-22-23-45-52+0100.txt](https://github.com/sriharshaarangi/BatteryChargeLimit/files/4100413/alogcat.2020-01-22-23-45-52%2B0100.txt)

Bagh33r4, 2020-01-26:
[image]

Same thing happens again this night, I plugget it in everything seemed normal, phone was at 78% limit set to 75%. When i woke up during the night all seemed Okey, phone was a bit higher at 80% not charging. In The morning it was at 100%.

Bagh33r4, 2020-01-26:
I believe that the phone activates charging every now and then for a secound, enough to increase the % if the phone is not in use. Sometimes I leave it plugged in while sharing wifi then this does not happen, charge limit stays below set level.

Any thoughts?

Bagh33r4, 2020-02-02:
Done more testing, my phone automatically starts charging every now and then even after turning it off with the batt_slate_mode file. When chargning is tuned off with this batt_slate_mode it goes into some maintain battery level mode, it seems to try to have the current draw be 0 going to or from the battery, so it only supplies the phone with enough power to maintain same battery level. I got it to work now, but not with the app, I made my own program in Tasker, which checks every 10 seconds if the phone is charging more then 200mA to the battery, if it is, it disables charging again in the batt_slate_mode file. But this was not enough, battery % still rises, so I installed an app that drains the battery and if the phones goes over 75% it starts the app and drains the battery back to 75% and closes the app.
Now I have left it on the charger for one night and it only goes to 75-76%.

milux, 2023-08-16:
> Done more testing, my phone automatically starts charging every now and then even after turning it off with the batt_slate_mode file. When chargning is tuned off with this batt_slate_mode it goes [...]

Interesting "solution", because intentionally drawing current and therefore causing additional cycling will likely do more harm than good to your battery. :see_no_evil: 
Some devices show such behavior, and it's ROM/charger-specific.
Funny enough, I found that on one of my devices showing similar behavior, changing the **charging cable** did the trick. Yeah, I know, strange, probably some weird charger behavior depending on the impedance of the cable...

closed 2023-08-16 by milux