sriharshaarangi/BatteryChargeLimit #187: Doesn't work on HTC 10

jigmasterj, 2020-03-06:
This app isn't stopping the battery charge at 80% on my rooted HTC 10. Please let me know what additional information I can provide to support. Thanks in advance!

Smithers90, 2020-03-08:
I switched to the control file "charger control", and it seems to be working as expected. I'll test it a little more but expect to be closing this issue soon.

Smithers90, 2020-03-10:
It's been working as designed with the correct setting, so that's great. It seems like if the software has attempted to disable the charger but it detects the battery is still charging, it could prompt the user to try a different charge setting. Just my thoughts for a potential way to reduce user confusion. I have no further comments other than "thanks for the great program!"

milux, 2023-08-16:
Since this issue is a little cluttered now, please open a new one with your precise feature wish if still on this.

closed 2023-08-16 by milux