sriharshaarangi/BatteryChargeLimit #193: android 10 issues

532910, 2020-09-02:
After upgrade to Android 10 (LineageOS 17.1) from Android 9 (LineageOS 16)
I've found my phone (OP3T) charged more than a limit, but only several times.
Battery optimization is off for BatteryChargeLimit.

milux, 2020-09-07:
I experienced similar issues with my Android 10 ROM. I observed that the responsible Broadcast is not triggered for certain chargers an my device. Appears to be an ROM issue or hardware bug. I think I have found a workaround, but I can't give you any ETA for a beta version with that right now, sorry...

532910, 2020-09-07:
Sure this is not a hardware bug in my case as I use the same original charges as before.

532910, 2020-09-07:
Moreover, I seen this issue only 3-5 times, though I charge phone.