sriharshaarangi/BatteryChargeLimit #204: Does not work on Google Pixel 4a with Android 11

mycephei, 2021-01-31:
Dont work...

RHSaliya, 2021-02-01:
Does not work for me too.

bu4sz, 2021-08-01:
Does not work

SootyOwl, 2021-09-17:
Using acc I found that this control file works.

`/sys/kernel/debug/google_charger/input_suspend, 0, 1`

bu4sz, 2021-09-18:
Thank you so much - this works!

Uhter, 2021-09-20:
> Using acc I found that this control file works.
> 
> `/sys/kernel/debug/google_charger/input_suspend, 0, 1`

Hello guys, I'm a newbie and I'm not sure what to do. Do I only have to fill "Control Data" in settings? Like this:

[image]

Or maybe the opposite: **0 as Enabled Value** and **1 as as Disabled Value**?

SootyOwl, 2021-09-20:
> Hello guys, I'm a newbie and I'm not sure what to do. Do I only have to fill "Control Data" in settings? Like this:
>
> [image]
>
> Or maybe the opposite: **0 as Enabled Value** and **1 as as Disabled [...]

Hi there. 0 is the enabled value, 1 is the disabled value. I've also had some issues if "Automatically reset battery stats" is turned on, but YMMV.

**Set path to desired file:** `/sys/kernel/debug/google_charger/input_suspend`
**Enabled value:** `0`
**Disabled value:** `1`

Uhter, 2021-09-20:
It didn't work at first time. Then I changed two other things:

**Enforce Charge Limit**: `unchecked`
**Always Write CTRL File**: `checked`

Now, it works!
Thank you @SootyOwl for you help ;)

Uhter, 2021-09-30:
@mycephei:

May be you can close this issue, no?

kaoneko, 2021-09-30:
Thanks for the info guys!

Changing the control file settings to `/sys/kernel/debug/google_charger/input_suspend, 0, 1` sort of worked BUT after reaching the charge limit my phone kept switching between charging and discharging at irregular intervals, e.g. 30 seconds, 15 seconds, 50 seconds (I kept an eye on the charge current in AccuBattery, Battery Charge Limit just keeps saying `Status: CHARGING` whether or not it actually is).

Unchecking _Enforce Charge Limit_ in Settings solved this (thanks @Uhter). I didn't need to check _Always Write CTRL File_.