sriharshaarangi/BatteryChargeLimit #205: Tasker and Battery Charge Limit

Shadeclan, 2021-02-10:
Is it possible to control Battery Charge Limit with Tasker?  If so, how would I do it?  In Tasker Tasks (Launch App), I can see the "LimitChangeActivity" but I don't know what data to send it to get it to activate / deactivate.  I can also see the "activities.MainActivity".

kaoneko, 2022-10-16:
You could just recreate the functionality of Battery Charge Limit in Tasker. Make a profile that activates when charging and charge is above the threshold. Enter task writes the "disabled" value to the control file (by executing a shell command as root, something like `echo 0 > controlfile`).

Shadeclan, 2022-10-20:
I looked at that and thought about doing that but in the end, I decided to uninstall Tasker altogether.  I found that profiles involving app monitoring drained my profile.  I found I could do what I wanted with [Autoroid](https://play.google.com/store/apps/details?id=com.soosu.autoroid) with a lot less battery drain.  As far as Battery Charge Limit is concerned, I rethought how and when I charge the phone and abandoned my Tasker idea.

milux, 2022-10-22:
> Is it possible to control Battery Charge Limit with Tasker? If so, how would I do it? In Tasker Tasks (Launch App), I can see the "LimitChangeActivity" but I don't know what data to send it to get it to activate / deactivate. I can also see the "activities.MainActivity".

Yes, you can, absolutely!
I recommend using a Broadcast for this, not the Intent/Activity way.
See the very README.md of this repository for directions! ;)

closed 2022-10-22 by milux

milux, 2022-10-22:
BTW: I'm controlling the limit via an Automate Flow.
https://mega.nz/file/AtdzCCDY#KH-pvO_TrtpiewfFsW8doZ-wsql7RSXrrpC-04OUrS0