sriharshaarangi/BatteryChargeLimit #207: Doesn't work with Huawei Y360-U31

Mike-0815, 2021-03-08:
Hello !

Installed BatteryChargeLimit, but unfortunately it doesn't work. It installs fine and I chose the Motorola Defy+ (MB526) control file.

As I use this phone nearly permanently on the power-cable, I choosed to hold a range of 50-55% battery level and started the service. When the battery level reached 55%, the app told me to hold 50-55% battery level. But the battery will be continously charged without limitations, like without the app installed.

Any help welcome, thanks in advance,
Kind regards
Mike

kagar4, 2021-03-14:
Try to use magisk module called input power control, May be work on your device

milux, 2023-08-16:
There is no guarantee whatsoever that BCL works with any device, or even specific ROM version.

closed 2023-08-16 by milux