sriharshaarangi/BatteryChargeLimit #210: Xiaomi Mi 11 support (found control file)

JisJ3rry, 2021-03-31:
TL;DR control file for Mi 11 is `/sys/class/qcom-battery/input_suspend, 0, 1`.

Hello, I spent the day trying to make the app work on my new Xiaomi Mi 11.
I couldn't find a working control file inside `/sys/class/power_supply`, so I initially thought suspending charging wasn't supported on this model.
In disbelief I kept on searching, and I found that the control file is situated in `/sys/class/qcom-battery/input_suspend`. The values obviously are 0 for enabled and 1 for disabled, since it's the usual Xiaomi input_suspend, just moved to a different location.
I don't know if this is a Snapdragon 888 thing or just a Mi 11 oddity, however if there's anyone else having problems with this app on Snapdragon 888, you could try looking in this path.

Could you update the app to reflect this? I know it's not mandatory since you can configure it yourself, but I ask just for completeness' sake. Thank you and thanks for your work on this awesome app.

SwiTool, 2021-05-18:
Thanks, as a Mi 11 user, I never searched the control file and you just saved me some precious time here!

0czko, 2022-09-03:
11t pro user here, miui 13/a12. Above fix works!

5ulo, 2023-08-29:
`/sys/class/qcom-battery/input_suspend, 0, 1` applies for Xiaomi 13 too

stefan2030, 2023-11-06:
> `/sys/class/qcom-battery/input_suspend, 0, 1` applies for Xiaomi 13 too

Also applies top Poco F5