sriharshaarangi/BatteryChargeLimit #211: Some features request

ioctl-user, 2021-04-03:
It would be nice if you add some useful features:

1. When charge limit was reached and device plugged in, use cable power instead of discharging battery. This feature can be ported from the Acc.
2. Implement charging voltage limits to prolong battery life.
3. Show charge/discharge current in addition to temperature and voltage.
4. Implement black theme and theme autoselection according to system settings.

danyougle, 2022-05-01:
Point 1 is what I'd like to say. It's crucial for my use case as a home server on charger.

milux, 2023-08-16:
1. Please be more specific about how this is done by ACC. This typically depends on the control file.
2. Planned, duplicate.
3. No Android-standard way of requesting that information, I think. Would be a device-specific improvement then. Please open separate PR if still interested.
4. Isn't that already available in the beta versions?

milux, 2023-08-16:
In general, please always open separate issues per feature, otherwise we cannot track anything efficiently.

closed 2023-08-16 by milux