sriharshaarangi/BatteryChargeLimit #215: Mojave support

yabo-boye, 2021-06-06:
Is there any way this can be supported on Mojave, even with limited functionality? I upgraded to Catalina just for this, unfortunately it has a few very annoying problems that are killing my workflow (especially Bluetooth issues). I would love some version of this to work in Mojave, and I would contribute to make it happen.

Thank you for your amazing work!

milux, 2021-06-08:
Sorry, I don't get it... we're building an Android application here, you're talking about MacOS versions... Please elaborate a little more, because I honestly have no idea what you are talking about, sorry. :sweat_smile:

yabo-boye, 2021-06-08:
oh shit. i got the wrong github, lol
i was meaning to post this in the forum that has a similar app for macos.
sorry for the confusion !
.
*∞ <3*
*+49 15258966399*

thejohnha, 2021-06-08:
@yabo-boye Now I'm curious - which is the other repo for macOS? thanks!

yabo-boye, 2021-06-08:
https://github.com/davidwernhart/AlDente

great app

closed 2021-06-09 by milux

thejohnha, 2021-06-09:
thanks @yabo-boye it's awesome!!! Now if I could just get this repo working on my android phone...