sriharshaarangi/BatteryChargeLimit #224: OnePlus Nord N200 5G battery control file

jimmyli97, 2021-10-04:
Thanks for writing a wonderful app! Here is the control file location for the Nord N200 5G: `/sys/devices/virtual/oplus_chg/battery/mmi_charging_enable`

JustRedTTG, 2022-12-18:
OMG, thanks so much. Confirmed works on OnePlus Nord 2T 5G

jimmyli97, 2025-06-04:
> How do I use it

Make sure your device is rooted and the app is installed per https://xdaforums.com/t/app-root-4-0-battery-charge-limit-v1-1-1.3557002/

In the settings, set the control file to the above file and that 0 is set to disable and 1 is set to enable

PrincessJasmine9, 2025-06-04:
Can't it work with root?

PrincessJasmine9, 2025-06-04:
> Your device needs to be rooted to use this app.
> If you don't have root, I would recommend downloading a battery charge
> alarm from the app store and disconnecting the charger by hand when the
> alarm goes off.

Ok thanks bro