sriharshaarangi/BatteryChargeLimit #227: Phone doesnt charge at all after installing this app

Jordan-Letchford, 2022-01-04:
Hi 

I love the idea of this app, so tried it on a rooted Galaxy S7. 

My phone battery was all good and everything worked before installing the app, after installing the app and setting the limits, and turning it on. My phone keeps losing charge, it says it's charging in the Battery charge limit app and my phone battery symbol confirms it is charging but the percentage just keeps dropping. Now my phone is flat and doesn't charge. Any idea what happened?

Cheers,
Jordan

Klimmbimmel, 2022-04-06:
Turn off your phone and load it then. That did help me (only loaded up to 75%)

milux, 2022-04-10:
Hi, I have experienced a similar thing with my Samsung Tablet. This is a ROM bug (feature?), and cannot be fixed on our side.
You can try another control file, if any. If that doesn't help, there's nothing we can do for you on our end. Sorry. :(

closed 2022-04-10 by milux

Klimmbimmel, 2022-08-30:
i fixed the issue using the app AccA from F-Droid (Advanced charging controller)