sriharshaarangi/BatteryChargeLimit #228: Pixel 6 rooted do not stop automatic

lavoie005, 2022-01-10:
Hi, i have rooted pixel 6 and the auto stop charging do not work on my phone

Pixel 6
Android 12
Version Number : SD1A.210817.036

thejohnha, 2022-01-17:
I don't know the methodology of how to figure out what to set, but I tried the settings that worked for my Pixel 4a and they did not work. The Pixel 4a's settings I used were:

Turn ON `Configurable CTRL File Data`
Click `Configure Control Data`
Set path to desired file:
/sys/kernel/debug/google_charger/input_suspend
Enabled value: 0
Disabled value: 1
Uncheck "Enforce Charge Limit"
Check "Always Write CTRL File"

tgice, 2022-02-07:
The default CTRL file of:

`Motorola Defy+ (MB526)`

also does not work for me.  It'd be great if anyone has figured this out yet and could post what to do!

closed 2023-08-16 by milux