sriharshaarangi/BatteryChargeLimit #253: Disconnects automatically

Gormar3, 2024-11-20:
Facing an issue where my s9 consistently turn this battery charge limit app off?app is literally running i am using mobile it starts charging itself even when its not below 41% approximately which is my limit.can someone tell me what is the issue?when i reopen the app or click it then it says battery charge limit enabled.so for some reason i think the limit is keep turning off by itself,app wont be put to sleep by system to save resources so thats not an issue.I dont know why its happening?literally when i am using mobile sometimes it happens and mobile starts charging ( when it should not have started )

Gormar3, 2024-11-23:
The problem is it stops.i open the app then its enabled again,as i have explained battery limiter does not work.it does nothing.does not stop charge it basically does nothing until i reopen the app or something.i hope this clarifies the issue.

Gormar3, 2024-11-23:
The app somehow is being stopped.i dont know why,the phone is suppose to not turn it off yet it apparently stops working itself.even if i have it opened in background.it randomly stops.its happening from sometime.maybe a mobile app or update is causing it i dont know.i did install an antivirus some months back,not sure if its related to that.but it started happening since last month or 2 so i dont think thats the issue.

milux, 2024-11-25:
OK, that's now surely something we can not provide any help with. There are millions of possible reasons why something on your phone is stopping BCL or let it crash.
You could try on XDA, maybe someone encountered the same issue and was able to solve it. Good luck.
Regarding the Antivirus (just personal opinion): I, personally, believe that Antivirus apps on an Android phone are rightout nonsense security theater. When your Antivirus requires root, it's highly dangerous itself. If it doesn't, it can't really do anything useful. All apps are sandboxed and can't do much harm unless you crack open this sandbox by giving an app root priviliges. If you don't trust an app and haven't checked its trustworthiness, never do that!
And when it comes to state-evel actors: No off-the-shelf protection will ever help you against exploit chains worth several million dollars. There is barely any effective protection against such attackers.

closed 2024-11-25 by milux